UCLA-Rocket-Project/OLD-Ares2022-2023 #4: Perform functional test of previous-year system.

harrisonCassar, 2022-10-11:
This will help us to familiarize ourselves with last year's/the current state of the system, and will also help motivate the setup of a "testbed"/system to test any functional changes on, which will be done in a separate issue.

This includes:
- Locate/setup hardware/subsystems.
- Setup local/ground station device (for connecting to Grafana server).
- Run a functional test, with and without hardware-in-the-loop.

Refer to Chris for any additional setup guidance.

harrisonCassar, 2022-10-28:
Functional tests (both with and without hardware) have been completed several times over:
- 10/17-10/21: of DAQ and Controls, with the Ground Systems team, just for development
- 10/25: of DAQ, with Ground Systems team, after some DAQ re-wiring changes
- 10/26: of DAQ, supporting PT calibrations
- 10/27: of DAQ and Controls, supporting leak checks
- 10/28: <to be completed>
- 10/29: <to be completed> of DAQ and Controls, supporting Cold Flow

Notable issues that deem changes in our codebase/various Software support tools have been properly ticketed: identified (potential) Software problems/TODOs:
- Re-sync RPi RTC (#6)
- Support for 9 PTs (#7)
- TC readings not appearing on Grafana (TODO)
- MPV switch improper mapping (#15)
- Creating of custom Grafana dashboard (with labels!) to support Cold Flow operations (#10)
- Investigate actual reading offsets despite apparent high-fidelity calibrations (TODO)
- Support easy means of code deployment to RPi (#16)

harrisonCassar, 2022-10-31:
Due to the several functional exercising of the Ground Systems electronics + software, this task has now been completed.

Passing off to @philipdoucla for concurrence before closure.

philipdoucla, 2022-11-01:
Concurred!

closed 2022-11-01 by philipdoucla