UCLA-Rocket-Project/OLD-Ares2022-2023 #6: Re-sync RTC module to current wall-time.

harrisonCassar, 2022-10-26:
# Description
Upon running a functional test, we discovered that the RTC module that the RPi uses to sync up to real-time every minute has drifted by about 20 seconds since last year (expected). This is not ideal, as all data that the RPi collects is timesynced with this time, and therefore when viewed by Grafana, there exists a visual "lag" in the datapoints due to all new data being timestamped 20 seconds in the past!

# Possible Fix
To fix this, we'll likely need to search up the RTC module's documentation for a process for doing this... Perhaps there's an Arduino library that has an API to allow us to set the time with a one-off Arduino script? I believe its the Adafriuit PiRTC DS3231: https://www.adafruit.com/product/4282

To verify this change, perform the re-time-sync, re-enable auto-timesyncing, restart the RPi, and start Grafana, verifying that all of our datastreams are at the current wall-time (screenshots required!).

# Current Workaround
A current workaround to this is to disable the RPi's RTC auto-syncing, and then to manually sync the time to the current UTC time as follows:
```bash
# Disable RPi's auto-syncing with RTC (0 to stop, 1 to start)
timedatectl set-ntp 0

# Manually set current UTC time!
timedatectl set-time HH:MM:SS
```

If we desire to re-enable the RPi's RTC auto-sycning, we then run the first command above (specifying 1 instead of 0). We may need to restart the RPi after this for it to take effect.

harrisonCassar, 2022-10-31:
Seemingly, by setting the time manually, the RTC module might have re-synced itself after being re-plugged back into the RPi...?

Leaving this issue open for now until evidence of this happening is posted.

philipdoucla, 2023-01-24:
Moved to Low Priority due to other more important work in pipeline (config files + deployment stuff #16).
Risk is low, due to fix happening. Still want to leave open though to investigate root cause.

~ TRC 1/23/23