UCLA-Rocket-Project/OLD-Ares2022-2023 #13: Consider use of polynominal best-fit to produce PT calibration values.

harrisonCassar, 2022-10-27:
# Description/Motivation
Currently, we take advantage of an assumption that our PTs behave linearly, utilizing a linear best-fit line to produce calibration values  `a,b` of the form `a*x + b` for a given raw voltage value of `x`. We've observed in the past that this results in perhaps non-zero produced PSI readings for ambient pressure/depressurized system (0 PSI), as well as perhaps lower-confidence in the outputted PSI readings for extreme PSI readings (close to minimum/maximum values rated for a specific PT).

# Proposed Solution
To fix this, we could perhaps consider a polynomial (or just non-linear) best-fit to produce calibration values for our PTs.

Some pros/cons to consider:
Pros:
- Could better capture non-linear behavior of PTs throughout entire operating range (perhaps most expected at extreme operating values).
- Avoids assumption that PTs' output react linearly to changes in pressure.
Cons:
- Could perhaps overfit to calibration data, which may have inherent noise that makes it stray from actual behavior on average.
- More computationally expensive (how much though? not sure until we test!)
- Strays from assumption that PTs are linear (as this is how they *should* behaving).
- Have to learn/compute an additional calibration value as a part of the calibration process.

Related to #10, #11.

harrisonCassar, 2022-10-31:
Labelling as low-priority due to reasonable assumption of linear-behaving PTs.

During operation, we'll be sure to be aware of this limitation for the time-being.

philipdoucla, 2023-01-24:
Through operations, PTs and LC demonstrated to be strongly linear.

At the current moment, linear best-fit proves to be the best option, and thus we move this issue to closure until future investigation.

~TRC 1/23/23

closed 2023-01-24 by philipdoucla