UCLA-Rocket-Project/OLD-Ares2022-2023 #9: Create updated Grafana dashboard to be used by Fall Quarter tests and operations.

harrisonCassar, 2022-10-27:
# Motivation/Description
Currently, we have a legacy Grafana dashboard that only showcases ~5 PT pressures. As noted in #7, Propulsion desires support for up to 9 PTs, and therefore we should ensure that we have a way of visualizing the current pressure values read by each of these sensors.

# Proposed Workflow to Update
Some proposed steps:
- Power the system
- Open the Grafana GUI
- Log in as Rocket Project Admin on Grafana (usr: rocket, psw: rocket)
- Go to the current dashboard ("Barracuda")
- Copy the current dashboard (name it, say, "Ares 2022-2023")
- Edit the dashboard as needed
    - Edit main PT panel to graph 9 input channels
    - Edit gauges below PT panel to show current value for all input channels
    - Add updated labels (speak to Prop for this)

harrisonCassar, 2022-10-28:
Current DRAFT for supporting 10/27-10/28 Leak Checks and Cold Flow:
[image]

Expected TODOs:
- Change gauge color thresholds based on expectations.
- Fully label all gauges.
- Separate certain "of interest" channels in different graphs/plots.
- Add constant "c-value" offset to specific channels (for ease of reading) in ADC source code.
- Re-name/re-label ADC data channels in ADC source code (re-configure dashboard to work with new names).

philipdoucla, 2023-01-24:
Closing due to passing of Fall 2022 test campaign.

Opening sister ticket #34 for Winter 2023 test campaign.

~ TRC 1/23/23

closed 2023-01-24 by philipdoucla