UCLA-Rocket-Project/OLD-Ares2022-2023 #31: Calibrate Load Cell for 11/5 Hotfire operations.

harrisonCassar, 2022-11-04:
# Motivation/Description
Currently, we do not have any recent calibration values for our Load Cell. We need a Load Cell (well-calibrated) for 11/5 Hotfire operations!

# Work To-Be-Done
Support Load Cell calibration, computing calibration coefficients, and updating ADC source code with results.

Similar to PT calibration process done in #10.

philipdoucla, 2022-12-12:
This will tentatively be done during the beginning of winter quarter with Emily and Logan (Electronics and VE Leads).

harrisonCassar, 2023-05-07:
This was completed but never closed. All work has been completed.

For completion sake, here's the complete calibration values (load cell is represented in the last column):

```
char labels[10][100] = {"lox_fill", "fuel_tank", "cc", "low_press", "lox_tank", "pneumatics", "lox_manifold", "fuel_manifold", "high_press", "load_cell"};
double a[10] = {245.29, 249.93, 247.55, 247.4, 248.81, 47.864, 251.65, 253.77, 1248.9, 462};
double b[10] = {-266.93, -271.7, -261.56, -274.58, -260.28, -61.123, -282.75, -272.97, -1233.3, 0.954};
```

closed 2023-05-07 by harrisonCassar